jgromes/LoRaLib #91: Porting to a non-arduino framework

ccvelandres, 2019-09-25:
Just a few queries about porting to a non arduino framework.

I browsed through the source code and found that all the classes for the specific chips inherits the Module Class which contains the methods for interfacing with the SPI driver from the arduino framework. 

1. If I need to port the library, I just need to replace the methods from the Module Class right? Specifically on Module::SPItransfer and the Module::init functions right?
2. the SPI mode here is half-duplex right?

jgromes, 2019-09-25:
1. Pretty much, all platform-dependent code should be in the Module class (let me know if that's not the case). Apart from what you already listed, you will also have to modify the constructor since it's using SPIClass.

2. Yes, SX127x chips do not send any information back during SPI transfer from master, so it's all just half duplex.

ccvelandres, 2019-09-27:
Got it to work on my device now. I just copy-pasted each function to make it easier to check if there's some things i needed to change and I had to drop the Physical Layer Class.

Closing this issue now.

closed 2019-09-27 by ccvelandres