jgromes/LoRaLib #92: softSpi configurations question/issue

enriqueromero, 2019-09-27:
Hello,
Newbie here, been on this for a while now.. I'm sure(hope) it's something really simple. I need to use the softSpi library. In code you stated the following:

```
// to use non-default SPI interface or software SPI
// implementation, pass the SPI interface to the LoRa
// constructor
//
// DigitalIO library for software SPI
// https://github.com/greiman/DigitalIO
#include <DigitalIO.h>       //uncommented
SoftSPI<4,5,6> softSpi; //uncommented

SX1278 softLora = new LoRa(7, 2, 3, softSpi );

void setup() {
  Serial.begin(115200);
   Serial.print(F("Initializing ... "));
  
  int state = lora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  }
  else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
```
I simply uncommented the lines above and error given thanks in advance:
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

Settings:64:45: error: no matching function for call to 'Module::Module(int, int, int, SoftSPI<12u, 11u, 13u>&)'

 SX1278 softLora = new LoRa(7, 2, 3, softSpi );//SPI

                                             ^

In file included from C:\.......\Documents\Arduino\libraries\LoRaLib-master\src/LoRaLib.h:20:0,

                 from C:\.....\AppData\Local\Temp\arduino_modified_sketch_873237\Settings.ino:28:

C:\....\Documents\Arduino\libraries\LoRaLib-master\src/Module.h:44:5: note: candidate: Module::Module(int, int, int, SPIClass&)

     Module(int cs = LORALIB_DEFAULT_SPI_CS, int int0 = 2, int int1 = 3, SPIClass& spi = SPI);

jgromes, 2019-09-27:
Seems DigitalIO library does support the SPIClass base, could you try with https://github.com/MajenkoLibraries/SoftSPI?

enriqueromero, 2019-09-27:
Thanks for you reply. Looks like it compiles fine with code below... but still not right. 

```
#include <SoftSPI.h>

SoftSPI softSpi(4, 5, 6);
SX1278 softLora = new LoRa(7, 2, 3, softSpi );//SPI

void setup() {
  Serial.begin(9600);
  Serial.print(F("Initializing ... "));
  
  int state = softLora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  }
  else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
```
I suspect something should be altered in module.h file could it be this line? how so? 
**Module(int cs = LORALIB_DEFAULT_SPI_CS, int int0 = 2, int int1 = 3, SPIClass& spi = SPI);**

jgromes, 2019-09-27:
> Looks like it compiles fine with code below... but still not right

OK, so what's the issue? Does the SPI not work? What's the error code?

enriqueromero, 2019-09-27:
SPI does not work. It's like it's still referencing the SPIClass from the standard SPI.h file and not the SoftSPI.h. I see signals on the standard pins(12,11,13) and not the pins(4,5,6) which I initialized at. 
Thanks.

jgromes, 2019-09-27:
Looks like SoftSPI doesn't implement `beginTransaction()` and `endTransaction()` methods from Arduino SPIClass, so the default SPI is used instead. I suggest you contact the author of that library to add those - unless the software driver is fully compatible with Arduino SPIClass, it won't work with LoRaLib (or, most likely, any other library that's using SPI).

The other option is to only use hardware SPI.

closed 2019-09-27 by jgromes