jgromes/LoRaLib #94: No packet received on receiver LoRa-02 . Everything worked perfectly fine before

Benedict-Bsl, 2019-11-28:
[image]

I am using the above lora module and the same connection with the example codes found in this interesting library(LoRALib by jgromes); with the "Transmit" and "Received" code provided. It was working perfectly fine before, but a few days ago, it is just stack at "Waiting for incoming transmission..." block. And it goes no further, while the receiver is actually sending real time data and all. 

Your help will greatly be appreciated. Thanks.

jgromes, 2019-11-28:
Seeing as you have the exact same issue using a different library (https://github.com/sandeepmistry/arduino-LoRa/issues/307), I suspect this is most likely a hardware issue, either on receiver, on transmitter, or both - though you haven't provided nearly enough information to even take a guess at what could be the cause

Assuming the photo is your actual setup, I doubt it will work long-term - Arduino Nano in that picture is connected directly to the LoRa module without any sort of logic level shift. Since Arduino Nano has 5V logic and SX127x modules have 3.3V logic, the LoRa module will get damaged eventually.

closed 2019-11-28 by jgromes

Benedict-Bsl, 2019-11-28:
I had the same problem with a different library, so i decided to use yours, i eventually ended up with the same issue. 
i have connected the LoRa module (as you can seen in the picture) to a LM317T regulator which outputs a voltage of 3.34V.

jgromes, 2019-11-28:
The supply voltage and logic voltage level are not the same: Arduino Nano outputs 5V signals. Those are connected directly to the LoRa module, which only has 3.3V inputs. This might work for some time, but eventually, the module will get damaged - exactly what seems to have happened in your case.

Benedict-Bsl, 2019-11-28:
I see what you mean. Any suggestions please? (never done that before to be honest).

Benedict-Bsl, 2019-11-28:
It will be of great help. Thank you.

Benedict-Bsl, 2019-11-28:
Can these 4 Channel Bi-Directional Logic Level Shifter Converter 3.3V-5V Logics Level Converter do the work?

jgromes, 2019-11-28:
Almost any shifter should do here, something like this will probably be fine: https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all

Alternatively, you could use 3.3V Arduino board. And you should probably change the LoRa module as well, hard to guess what exactly is broken with the current one.

Also, you should probably use RadioLib instead of LoRaLib if you want long term support, since LoRaLib will be discontinued on the near future.

Benedict-Bsl, 2019-11-28:
Thank you so much. I bought one of these a long time ago, and i never thought of using them. Thank you so much again.