jgromes/LoRaLib #96: Setting sync word - incorrect registers?

sjd227, 2020-02-14:
I think there's a bug setting the sync word to the correct registers. 

In sx127x.cpp, under `int16_t SX127x::setSyncWord(uint8_t* syncWord, size_t len)`, it looks like all bytes of the sync word are only written to RegSyncValue1 (line 804):

`  _mod->SPIwriteRegisterBurst(SX127X_REG_SYNC_VALUE_1, syncWord, len);`

However, each byte of `syncWord` should be written to `RegSyncValue1` through `RegSyncValue8` respectively (see screenshot of sx1272 datasheet below). 

[image]

jgromes, 2020-02-14:
The method used there is `SPIwriteRegisterBurst` - it will write in "burst" mode, i.e. write `len` bytes of data to `len` registers, starting at `SX127X_REG_SYNC_VALUE_1`. It's more convenient and faster than calling `SPIwriteRegisterBurst` multiple times.

closed 2020-02-14 by jgromes

sjd227, 2020-02-14:
Got it - thanks!