jgromes/LoRaLib #97: Question about preambles

sjd227, 2020-02-14:
I would like my sx1272 to receive packets having a specific preamble (111110010101001100000; 21 symbols), but I don't see anywhere in the library that allows this, other than just specifying the length (for which I'm setting the `PreambleLength` register to 17 such that 17 + 4 = 21).

Is it not necessary to set a specific preamble? I.e., will the PreambleDetector always pick up this preamble if I simply set the PreambleLength to 17? 

Thanks,
Steven

jgromes, 2020-02-14:
Are you using FSK or LoRa modem? Preambles work very differently between the two.

sjd227, 2020-02-14:
Im configuring FSK and enabling OOK (the signals I'm looking for are OOK, Manchester encoded).

jgromes, 2020-02-14:
I think SX127x can oly set the preamble to 0xAA or 0x55, so I'm not sure it will be able to receive those packets at all. Also, the preamble length is in bytes, not in bits. The "+4 symbols" only applies to LoRa preamble.

I don't think SX127x is the right radio to use here, though it could be possible to use the direct mode and try sampling the output on DIO1/DIO2 pins.

sjd227, 2020-02-14:
Hm ok, what do you mean by "direct mode"? 

Could I turn off the Preamble Detector and search for that preamble manually in the FIFO register?

jgromes, 2020-02-14:
> Could I turn off the Preamble Detector

Not sure about that, the demodulator needs the preamble to lock onto the packet, otherwise it doesn't know where each bit starts and ends.

SX127x allows you to skip the packet operations and directly output demodulated data on DIO1 (clock) and DIO2 (data) pins. I haven't tried to receieve data this way, but it's the only way I can think of SX127x *could* be able to receive those packets.

How do you know the preamble is 111110010101001100000? Preamble is usually a sequence of alternating bits.

sjd227, 2020-02-14:
Ok, what do you think about treating the preamble as a "sync word" to lock in the packet, keeping the Preamble Detector off? 

> SX127x allows you to skip the packet operations and directly output demodulated data on DIO1 (clock) and DIO2 (data) pins. I haven't tried to receieve data this way, but it's the only way I can think of SX127x could be able to receive those packets.

Thanks - I will look into this. 

> How do you know the preamble is 111110010101001100000?

I am looking for "Encoder Receiver Transmitter" (ERT) messages; it's used by utility meters to report consumption data (e.g., electricity, gas, water). The preamble is listed on the [wikipedia page](https://en.wikipedia.org/wiki/Encoder_receiver_transmitter) (which is the above preamble with the "sync bit" added to the front). I saw [another project](https://github.com/bemasher/rtlamr) use this preamble as well for ERT messages.

jgromes, 2020-02-14:
I really don't know what will happen when you turn off preamble detector, sorry. I haven't tried running SX127x in that mode (which is also why the library doesn't expose preamble detector to the public API). Guess you'll have to try it out and see ;)

Btw - If you're looking to create an ERT decoder, I suggest you use [RadioLib](https://github.com/jgromes/RadioLib) instead of LoRaLib. RadioLib is using the same SX127x driver as LoRaLib, but also allows you to implement higher-level protocols - it already has support for RTTY, Morse code and AX.25.

Besides, LoRaLib will be discontinued in the near future.

closed 2020-02-14 by jgromes

jgromes, 2020-02-16:
@sjd227 I just noticed CC1101 (one of the modules supported by RadioLib) allows to skip preamble synchronization - perhaps that could be useful to you?

sjd227, 2020-02-20:
@jgromes thanks for the input. I looked at CC1101 - it seemed pretty similar to turning off the preamble detector on the SX1272 (by setting `SX127X_PREAMBLE_DETECTOR_OFF` to the `SX127X_REG_PREAMBLE_DETECT` register, from my understanding). 

So now I'm trying to decode IDM packets (another [ERT protocol](https://en.wikipedia.org/wiki/Encoder_receiver_transmitter)) as a proof-of-concept, since they use a standard preamble (0x5555).

It specifies the "frame sync" as 0x16A3. 

Do you know if "frame sync" is the same as the "sync word" that can be set by `SX127x::setSyncWord(uint8_t* syncWord, size_t len)`?

I'll make sure to use RadioLib from now on!

jgromes, 2020-02-21:
SX127x FSK sync word is the first 1 - 8 bytes after preamble, so in theory 0x16A3 as the sync word should work.