jgromes/LoRaLib #98: cannot get the ARRAY transmit and receive working.

prerak9, 2020-03-18:
Hello,
Thanks for your amazing work.
I am not able to get the array to transmit and receive working, can you please share an example code for transmitting and receiving array.
Thank you

jgromes, 2020-03-18:
What does "not working" mean? Does not compile/nonsense output/crash ... Please share the exact code you're running, as well as the output from serial monitor, library version, Arduino IDE version and which Arduino board are you using.

Also, LoRalib will be discontinued in the near future (see [EOL](https://github.com/jgromes/LoRaLib/blob/master/EOL.md) for details), use [RadioLib](https://github.com/jgromes/RadioLib) instead.

prerak9, 2020-03-18:
Hello,
Is the receive heap crash solved in RadioLib because i was struggling with
it on loraLib.

thank you

jgromes, 2020-03-18:
Most likely, though I can't tell you for sure since you haven't provided any more information. As far as I know, there should be no memory issues in the latest version of LoRaLib or RadioLib.

prerak9, 2020-03-18:
Actually i am using LoraLib with example receive and transmit code where
after few hours of run the receiver hangs up and only solution is to reset
the board to get it back running whereas transmitter works perfect.

jgromes, 2020-03-18:
> Please share the exact code you're running, as well as the output from serial monitor, library version, Arduino IDE version and which Arduino board are you using.

I really need the above information. Otherwise any attempt at debugging is just a guesswork - what modifications did you do to the examples?

prerak9, 2020-03-18:
its the EXACT code from your example with no changes at all.
and there is no serial monitor output but the receiver hangs up and no data
is received after few hours

prerak9, 2020-03-18:
also here on of my relay switch code with also tends to hangs up.

receiver.......

#include <RadioLib.h>

SX1278 lora = new Module(10, 2, 9, 3);
int k = 0;
int state;

 String str = "0";
void setup() {
  Serial.begin(9600);
pinMode(4,OUTPUT);
pinMode(5,OUTPUT);
pinMode(6,OUTPUT);
pinMode(7,OUTPUT);
pinMode(8,OUTPUT);
pinMode(A3,INPUT);
pinMode(A5,INPUT);
  digitalWrite(5,HIGH);
  digitalWrite(6,HIGH);
  digitalWrite(7,HIGH);
   digitalWrite(8,HIGH);

  delay(1000);

  digitalWrite(8,LOW);
  digitalWrite(5,LOW);
digitalWrite(6,LOW);
digitalWrite(7,LOW);
  state = lora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
    state = lora.setFrequency(436);
  state = lora.setOutputPower(20.0);
  state = lora.setCurrentLimit(200);
  lora.sleep();
}

void loop() {

if(str=="1"&& k ==0){
   digitalWrite(5,HIGH);
  digitalWrite(7,HIGH);
  digitalWrite(6,HIGH);

  digitalWrite(8,LOW);
  digitalWrite(4,LOW);
   str = "0";
  delay(100);

}

else if(str=="2"&& k ==0){
   digitalWrite(7,LOW);
  digitalWrite(6,HIGH);
  digitalWrite(5,HIGH);

  digitalWrite(8,LOW);
 str = "0";
  delay(100);
}

else if(str=="3"&&k==0){
   digitalWrite(7,LOW);
  digitalWrite(6,LOW);
  digitalWrite(5,HIGH);

  digitalWrite(8,LOW);
  digitalWrite(4,HIGH);
  str = "0";
  delay(100);

}

else if(str=="4"&&k==0){
   digitalWrite(7,LOW);
  digitalWrite(6,LOW);
  digitalWrite(5,LOW);

  digitalWrite(8,LOW);
  digitalWrite(4,HIGH);
   str = "0";
  delay(100);

}
if(digitalRead(A5)==1 && digitalRead(A3)==1){

    digitalWrite(8,LOW);
}
else if(digitalRead(A5)==0 && digitalRead(A3)==0){

   digitalWrite(8,HIGH);
    digitalWrite(4,LOW);
}

 lora.receive(str);
delay(50);
}

transmitter.......

/*
   LoRaLib Transmit Example

   This example transmits LoRa packets with one second delays
   between them. Each packet contains up to 256 bytes
   of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   For more detailed information, see the LoRaLib Wiki
   https://github.com/jgromes/LoRaLib/wiki

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/LoRaLib/
*/
#include "LowPower.h"

#include <RadioLib.h>
int a = 0;
int b = 0;
int c = 0;
int d = 0;
int e = 0;
// create instance of LoRa class using SX1278 module
// this pinout corresponds to RadioShield
// https://github.com/jgromes/RadioShield
// NSS pin:   10 (4 on ESP32/ESP8266 boards)
// DIO0 pin:  2
// DIO1 pin:  3
SX1278 lora = new Module(10, 2, 9, 3);
String msg = "";
void setup() {
  Serial.begin(9600);
 pinMode(6, INPUT);
  pinMode(5, INPUT);
  pinMode(7, INPUT);
  pinMode(8, INPUT);
  // initialize SX1278 with default settings
  Serial.print(F("Initializing ... "));
  // carrier frequency:           434.0 MHz
  // bandwidth:                   125.0 kHz
  // spreading factor:            9
  // coding rate:                 7
  // sync word:                   0x12
  // output power:                17 dBm
  // current limit:               100 mA
  // preamble length:             8 symbols
  // amplifier gain:              0 (automatic gain control)
  int state = lora.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
 state = lora.setFrequency(436);
  state = lora.setOutputPower(20.0);
  state = lora.setCurrentLimit(200);
}

void loop() {
   lora.sleep() ;
  delay(50);
   LowPower.powerDown(SLEEP_2S, ADC_OFF, BOD_OFF);
  LowPower.powerDown(SLEEP_2S, ADC_OFF, BOD_OFF);
   LowPower.powerDown(SLEEP_2S, ADC_OFF, BOD_OFF);
  LowPower.powerDown(SLEEP_4S, ADC_OFF, BOD_OFF);
  LowPower.powerDown(SLEEP_4S, ADC_OFF, BOD_OFF);
  LowPower.powerDown(SLEEP_4S, ADC_OFF, BOD_OFF);
  LowPower.powerDown(SLEEP_4S, ADC_OFF, BOD_OFF);

  delay(50);

  if (digitalRead(5) == 0 && digitalRead(6) == 0 && digitalRead(7) == 0 &&
d==0) {

msg = "1";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
    a = 0;
    b = 0;
    c = 0;
    d = 1;
    e = 0;
    }

    if (digitalRead(5) == 1 && digitalRead(6) == 0 && digitalRead(7) ==
0&&c==0) {

      msg = "2";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
    a = 0;
    b = 0;
    c = 1;
    d = 0;
    e=0;
    }

  if (digitalRead(5) == 1 && digitalRead(6) == 1 && digitalRead(7) == 0 &&
b==0) {

 msg = "3";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
 a = 0;
    b = 1;
    c = 0;
    d = 0;
    e = 0;
  }

  if (digitalRead(5) == 1 && digitalRead(6) == 1 && digitalRead(7) == 1
&&a==0) {

msg = "4";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
    a = 1;
    b = 0;
    c = 0;
    d = 0;
    e = 0;
  }
  if(e==17){
      if (digitalRead(5) == 0 && digitalRead(6) == 0 && digitalRead(7) == 0
) {

msg = "1";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);

    e = 0;
    }

    if (digitalRead(5) == 1 && digitalRead(6) == 0 && digitalRead(7) == 0) {

     msg = "2";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
    e=0;
    }

  if (digitalRead(5) == 1 && digitalRead(6) == 1 && digitalRead(7) == 0) {

msg = "3";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
    e = 0;
  }

  if (digitalRead(5) == 1 && digitalRead(6) == 1 && digitalRead(7) == 1 ) {

msg = "4";
delay(60);
lora.transmit(msg);
delay(60);
lora.transmit(msg);
delay(60);
    e = 0;
 }
  }
  e=e+1;
   delay(50);
}

jgromes, 2020-03-18:
What about the library version, Arduino IDE version and which Arduino board is used? 

Also, if there's no output, how do you know heap crash is responsible? And how do you know transmitter is still transmitting - was that verified somehow? Are you using the blocking receive example or interrupts?

jgromes, 2020-03-18:
Please don't email code, it can't be fromatted using Markdown and that makes it neraly impossible to read ...

prerak9, 2020-03-18:
yes i verified the working of the transmitter by using another receiver
board which was responding to the signals.
i am using atmega 328 standalone on perf board. i am using the basic
receive example because the interrupt receive only receives one signal and
hangs up until i reset my mcu.

prerak9, 2020-03-18:
String str;
int state = lora.receive(str);

this is the function i am using to receive the data.
i am using the latest LoraLib and latest arduino ide.

jgromes, 2020-03-18:
I can't seem to replicate the original issue (array transmission "not working" - whatever that means): on Arduino Uno with a couple of SX1278 chips, transmitting and receiving array works fine.

Regarding the second issue with receiver hanging after a couple of hours - I'll keep the setup running for today and check on it occasionaly.

prerak9, 2020-03-18:
Thanks alot for your support. Kindly update me with status of your setup
once done testing:)

jgromes, 2020-03-18:
It's been running at 5 hours now with no issues whatsoever, receiving one packet per second - if this was indeed a memory issue I would expect it to crash within 30 minutes. This seems to suggest some problem in hardware.

Closing as I can't reproduce either issue.

closed 2020-03-18 by jgromes

prerak9, 2020-03-18:
Ok thanks for the update.
Last thing i want to ask is....
I have 10 pair of lora module 2 of each communicating with each other, so
which settings should i do to avoid them clashing with each other.

Code is the same receive and transmit in example.

jgromes, 2020-03-18:
Probably the best solution (without implementing some network protocol) would be to split your frequency band into channels, so that each pair has its own channel. Assuming you're using 125 kHz bandwidth, 200 kHz channel spacing should give you more than enough safety margin between adjacent channels. For example, one pair could be at 433.1 MHz, another at 433.3 MHz and so on. Remember to check which frequencies are legal to use without licenswe in your country.

prerak9, 2020-03-18:
Yes...its working this way....thanks alot sir for you very kind cooperative
response:)